zocker-160/keyboard-center #77: Some icons missing from GUI

aurtzy, 2025-07-26:
## Background

In case the background might be useful information: I recently managed to fix an issue I've had with missing icons on GNOME (unrelated to [#55](https://github.com/zocker-160/keyboard-center/issues/55)).  This appears to be due to Adwaita no longer supporting the [Freedesktop Icon Naming Specification](https://specifications.freedesktop.org/icon-naming-spec/latest/) due to various reasons including moving away from fullcolor icons, and was fixed by installing the `adwaita-icon-theme-legacy` package (see [this](https://discourse.gnome.org/t/adwaita-icon-theme-legacy-release/21021) and [this](https://gitlab.gnome.org/GNOME/adwaita-icon-theme/-/issues/288) for more information).

I think `adwaita-icon-theme-legacy` is usually included by default with GNOME/Adwaita on other distributions; I'm on a more niche distribution which happened to not have it, and will be looking into suggesting changes downstream, so I'm not sure if there's anything actionable here in this regard.

## Issue

After resolving the issue mentioned above, I noticed there were a few icons still not showing up.  It looks like these missing icons are not listed as part of the Freedesktop specification:

-   `document-edit`, for "edit LUA script".
-   `dialog-ok`, for "validate LUA script".
-   `chronometer`, for "add delay".

Possible suggestions for fitting alternatives I could find after a perusal of the specification might be `accessories-text-editor`, `system-search`, and `media-seek-forward`, respectively.

WDYT?

zocker-160, 2025-07-26:
Thank you for this report.

I will add a note to the readme, there is not much else what I can do sadly.

This entire icon problem is so incredibly frustrating, what is even the point of a freedesktop standard when it is just fuck all anyway, it is such a pain.

Thank you for trying to find alternatives, sadly it is not that easy.

For the "edit LUA script" I want to have a pen. 
`accessories-text-editor` is sadly a page icon for me. `draw-freehand`, but that also is not in the list.

`system-search` sadly gives a magnifying glass for me, which I don't like either, I want a check type icon.

For `add delay` I have found `appointment-new`, I am happy with that.

The reason why I haven't created my own icons is mainly so that it blends as well as possible to custom themes of users. Also reliably checking if user has enabled dark mode and then inverting the colors is also a pain, because there is no good standard API to check that.

I feel like though that finding fitting default icons has already costed me more time than creating my own.....

zocker-160, 2025-07-27:
Does `edit-symbolic` work for you?

aurtzy, 2025-07-27:
> Thank you for the report.
> 
> I will add a note to the readme, there is not much else I can do sadly.
> 
> This entire icon problem is so incredibly frustrating, what is even the point of a freedesktop standard when it is just fuck all anyway, it is such a pain.

Yeah, I empathize - the adwaita-icon-theme issue discussion I linked seemed to indicate a lot of complexities surrounding it.

> Thank you for trying to find alternatives, sadly it is not that easy.
> 
> For the "edit LUA script" I want to have a pen. `accessories-text-editor` is sadly a page icon for me. `draw-freehand`, but that also is not in the list. Maaaaybe `edit-copy`, but that might be misleading.
> 
> `system-search` sadly gives a magnifying glass for me, which I don't like either, I want a check type icon.
> 
> For `add delay` I have found `appointment-new`, I am happy with that.
> 
> The reason why I haven't created my own icons is mainly so that it blends as well as possible to custom themes of users. Also reliably checking if user has enabled dark mode and then inverting the colors is also a pain, because there is no good standard API to check that.

I see...

> Does `edit-symbolic` work for you?

No, that doesn't appear to exist in Adwaita. If you're okay with symbolic icons though, there is `document-edit-symbolic`.

aurtzy, 2025-07-28:
A note that was apparently not included in my last reply: `document-edit-symbolic` seems to exist for both Breeze and Adwaita, although it's not part of the specification either...